laurb9/tiny_scope #23: Unable to get this working on an Uno R3, or a Pro Mini. Seems to 'somewhat' work on a Leonardo

jkittle, 2018-09-09:
I'm hoping you can point me in the right direction - I'm trying to get this working on a 328P based atmega (to some extent, they're all the same for my purposes), but having issues. When the main module tries to run success= to fetch the analog values, I'm always getting a 0 returned, and an 'out of memory' error. I suspect the ADC reading is what's failing, but I'm not yet familiar enough with this to be able to debug it myself.  Is this really capable of running on an Uno/Micro, or has it evolved to where you're doing things only the Teensy can handle?

jkittle, 2018-09-09:
Some more information.   If I change the display type to 128x32 it samples fine (i mean, obviously the display is messed up).  It's only giving me the out of memory error on a 128x64 OLED.

laurb9, 2018-09-09:
Should be still working on AVR. Memory usage was tight but it was working, the Teensy has its own branch. I suspect changes in the compiler flags or improvements in the Adafruit SSD1306 library might have increased the size.

Do you get memory error during compile, or at runtime ? I didn't think it can throw an error, it usually just hangs when it ran out of space

jkittle, 2018-09-09:
It compiles just fine, I’m only getting the error at runtime.

jkittle, 2018-09-12:
Any chance you have time to look at this? I really want to use this
functionality in an upcoming project. Thanks!  I'm happy to test/help how I
can.

laurb9, 2018-09-24:
Have not had a chance to find an Uno to try with yet. I think if it's an issue with newer more bloated code that finding an older lib for the ssd might also work. For reference, this is what my compile output says (I haven't actually run it):

```
Using library EEPROM at version 2.0 in folder: ...
Using library Adafruit_SSD1306 at version 1.1.2 in folder: ...
Using library SPI at version 1.0 in folder: ...
Using library Adafruit_GFX_Library at version 1.1.5 in folder: ...
Using library Wire at version 1.0 in folder: ...
Sketch uses 18776 bytes (58%) of program storage space. Maximum is 32256 bytes.
Global variables use 1508 bytes (73%) of dynamic memory, leaving 540 bytes for local variables. Maximum is 2048 bytes.
```

laurb9, 2018-10-01:
I've tried it on a UNO clone and it seems to be running fine:

[image]

jkittle, 2018-10-01:
What version of libraries and which version of the Arduino IDE ?

laurb9, 2018-10-02:
Arduino 1.8.7 and the libraries I listed above.

laurb9, 2019-03-30:
Closed, assumed resolved

closed 2019-03-30 by laurb9