laurb9/tiny_scope #24: No image on 128x64 display

Radiofreak1041, 2018-12-25:
I'm using an Arduino Nano to drive the I2C OLED display. The code works fine when I select 128x32 in the Adafruit_SSD1306 library, but then I get incorrectly scaled text. When I have the 128x64 line enabled, the display remains black. Other projects work fine when using the Arduino Nano and the 128x64 resolution. Does this have anything to do with the memory size of the Arduino?

laurb9, 2018-12-27:
I believe so, as this problem was reported earlier (in #23 ). I suspect that one of the updated libraries now uses more memory (it was pretty close before).

I've tested it and it still works with Arduino 1.8.7 and the (older) libs below:
```Using library EEPROM at version 2.0 in folder: ...
Using library Adafruit_SSD1306 at version 1.1.2 in folder: ...
Using library SPI at version 1.0 in folder: ...
Using library Adafruit_GFX_Library at version 1.1.5 in folder: ...
Using library Wire at version 1.0 in folder: ...
Sketch uses 18776 bytes (58%) of program storage space. Maximum is 32256 bytes.
Global variables use 1508 bytes (73%) of dynamic memory, leaving 540 bytes for local variables. Maximum is 2048 bytes.
```

Radiofreak1041, 2018-12-28:
Thanks! It works now on the Arduino Nano. Didn't try that one.

closed 2018-12-28 by Radiofreak1041